scuttlerobot/SCUTTLE #11: BBB system space Issue

SantiagoGarza96, 2019-03-07:
[image]

I get this Issue when I try to use apt-get install "program name" to download programs. Any idea on checking and managing the BBB system storage.

ansarid, 2019-03-07:
Try running these commands:

```
cd /opt/scripts/tools
git pull
sudo ./grow_partition.sh
sudo reboot
```

SantiagoGarza96, 2019-03-09:
Thank you very much it worked. If possible can you explain what this did?

closed 2019-03-09 by ansarid

dmalawey, 2019-03-09:
Daniyal found the fix, not me but I just googled “Linux grow_partition” and found this:

https://elinux.org/Beagleboard:Expanding_File_System_Partition_On_A_microSD

ansarid, 2019-03-10:
@SantiagoGarza96 Sure!

```
cd /opt/scripts/tools
```
Moves you into the ```/opt/scripts/tools``` directory.
```
git pull
```
Since the directory you are in is a git repository that comes already downloaded to the BeagleBone it may be out of date. ```git pull``` updates the repository. When you run the command you will see the link https://github.com/RobertCNelson/boot-scripts show up since that is the original location of the repository.
```
sudo ./grow_partition.sh
```
This command runs the shell script ```grow_partition.sh``` which you can open and read through but basically gets the size of the sd card it is on and then expands the root partition to fill that space.
```
sudo reboot
```
Reboots the BeagleBone.