scuttlerobot/SCUTTLE #12: Find cause and resolution for "signal handler error"

dmalawey, 2019-03-08:
This output is given in concurrence with a disconnection of the beaglebone and a non-responsiveness to further commands.

/usr/local/lib/python3.5/dist-packages/rcpy-0.5.0-py3.5-linux-armv7l.egg/rcpy/__init__.py:127: UserWarning: > Installing signal handlers
  warnings.warn('> Installing signal handlers')

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.327422] Internal error: : 1028 [#1] PREEMPT SMP ARM

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.457984] Process python3 (pid: 1141, stack limit = 0xd1816218)

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.464104] Stack: (0xd1817e18 to 0xd1818000)

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.468482] 7e00:                                                       00000001 00000000

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.476700] 7e20: 00000000 d1817e7c db49b500 c081df48 db49b520 00000000 00000000 dc7d9210

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.484917] 7e40: d1817e74 d1817e50 c081b72c c081df54 c1504dc8 00000000 dc4e6f94 db49b500

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.493134] 7e60: 00000002 dc5f3280 d1817eb4 d1817e78 c081be7c c081b618 00000000 00009c40

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.501351] 7e80: 00000000 00000000 00000001 322c3fc2 00000002 c081be00 dc7d9200 00000000

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.509568] 7ea0: 00000000 d1817f68 d1817ecc d1817eb8 c092f108 c081be0c c092f0e0 dc7d9200

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.517786] 7ec0: d1817ee4 d1817ed0 c03855c4 c092f0ec 00000002 dc7d9200 d1817f1c d1817ee8

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.526004] 7ee0: c0384c10 c0385580 00000000 00000000 db702988 c0384b18 dc4f3840 a6e18aec

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.534222] 7f00: d1817f68 00000000 a6e18aec 00000002 d1817f34 d1817f20 c02f9258 c0384b24

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.542439] 7f20: 00000002 dc4f3840 d1817f64 d1817f38 c02f943c c02f923c c031b088 c031bbd0

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.550655] 7f40: c1504dc8 dc4f3843 00000000 00000000 dc4f3840 a6e18aec d1817fa4 d1817f68

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.558872] 7f60: c02f96a4 c02f9394 00000000 00000000 00000002 322c3fc2 00000005 00000001

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.567089] 7f80: 00000010 00000000 00000004 c01090e4 d1816000 00000000 00000000 d1817fa8

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.575306] 7fa0: c0108f00 c02f9654 00000001 00000010 00000009 a6e18aec 00000002 a6e39620

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.583522] 7fc0: 00000001 00000010 00000000 00000004 beccb290 0000000b 0000000c 0000000d

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.591739] 7fe0: 00000000 beccb1d0 00000000 b6f303d0 000b0030 00000009 00000000 00000000

Message from syslogd@scuttle at Mar  1 22:46:44 ...
 kernel:[  649.670935] Code: e5980000 e6ffa07e eb049325 e5985028 (e1d570b0)

ansarid, 2019-03-12:
[rcpy Issue #14](https://github.com/mcdeoliveira/rcpy/issues/14#issue-371687150)

dmalawey, 2019-03-13:
I have documented this issue in the Beaglebone.org beagleboard - blue forum.   [Link is here](https://beagleboard.org/discuss?place=msg%2Fbeaglebone%2FcdywN2pd0_E%2FVkOEB_4MAgAJ)

dmalawey, 2019-03-13:
I have also pinged the RCPY author on his library's [existing issue](https://github.com/mcdeoliveira/rcpy/issues/14), with the same topic.

dmalawey, 2019-03-21:
I've now added this as an issue at [StrawsonDesign/LibRobotControl](https://github.com/StrawsonDesign/librobotcontrol/issues/154#issue-423478005) which I am told is the library truly handling this code, since RCPY is a python-written reference to librobotcontrol.

dmalawey, 2019-03-26:
The issue is being discussed in the slack channel for libRobotControl [here.](https://librobotcontrol.slack.com/messages/CBLL51BTL/)

User jadonk is creating some troubleshooting scripts associated with the errror [here](https://gist.github.com/jadonk/8ece4ad83ac67e5215af6a3d686d0ae2)

dmalawey, 2019-03-27:
A fix is in the works.  All users should update their librobotcontrol library, after running sudo apt-get update.

An example is shown below.  You can confirm that you have the latest librobotcontrol version by attempting to update it again, and receiving feedback that the library "is already the newest version"

```
debian@scuttle:~$ sudo apt-get update
[sudo] password for debian:
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Get:3 http://repos.rcn-ee.com/debian stretch InRelease [3,061 B]
Get:4 http://deb.debian.org/debian-security stretch/updates InRelease [94.3 kB]
Hit:5 http://deb.debian.org/debian stretch Release
Get:6 http://repos.rcn-ee.com/debian stretch/main armhf Packages [912 kB]
Get:7 http://deb.debian.org/debian-security stretch/updates/main armhf Packages [466 kB]
Fetched 1,567 kB in 25s (61.2 kB/s)
Reading package lists... Done
debian@scuttle:~$ sudo apt-get install librobotcontrol
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be upgraded:
  librobotcontrol
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 399 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian stretch/main armhf librobotcontrol armhf 1.0.4-git20190227.0-0rcnee0~stretch+20190326 [399 kB]
Fetched 399 kB in 0s (828 kB/s)
Preconfiguring packages ...
(Reading database ... 53944 files and directories currently installed.)
Preparing to unpack .../librobotcontrol_1.0.4-git20190227.0-0rcnee0~stretch+20190326_armhf.deb ...
Removed /etc/systemd/system/multi-user.target.wants/rc_battery_monitor.service.
Removed /etc/systemd/system/multi-user.target.wants/robotcontrol.service.
Unpacking librobotcontrol (1.0.4-git20190227.0-0rcnee0~stretch+20190326) over (1.0.4-git20190107.0-0rcnee0~stretch+20190108) ...
librobotcontrol Package Uninstallation Complete
Setting up librobotcontrol (1.0.4-git20190227.0-0rcnee0~stretch+20190326) ...
Installing new version of config file /etc/modules-load.d/robotcontrol_modules.conf ...
Created symlink /etc/systemd/system/multi-user.target.wants/robotcontrol.service → /lib/systemd/system/robotcontrol.service.
Created symlink /etc/systemd/system/multi-user.target.wants/rc_battery_monitor.service → /lib/systemd/system/rc_battery_monitor.service.
leaving current robot control startup program alone
**********************************************
librobotcontrol package installation complete.
**********************************************
Processing triggers for libc-bin (2.24-11+deb9u4) ...
debian@scuttle:~$ sudo apt-get install librobotcontrol
Reading package lists... Done
Building dependency tree
Reading state information... Done
librobotcontrol is already the newest version (1.0.4-git20190227.0-0rcnee0~stretch+20190326).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
debian@scuttle:~$

```

ansarid, 2019-03-28:
[The issue has been patched.](https://github.com/rcn-ee/librobotcontrol/commit/06c406de2c7b63a246a525f243d4e170ec0c664d)

Run:
```
sudo apt-get update
sudo apt-get install librobotcontrol
```

closed 2019-03-28 by ansarid