scuttlerobot/SCUTTLE #14: Share skeleton code for Speech-to-Text

dmalawey, 2019-03-10:
Please include:

1. Libraries installed (or libraries referenced, which we already installed)
2. Links to the API or documentation
3. Commands or minimal code to get started
4. Description of hardware used to accomplish the task

dmalawey, 2019-03-10:
[Trial adding JASPER to Beaglebone.docx](https://github.com/dmalawey/Scuttle/files/2950171/Trial.adding.JASPER.to.Beaglebone.docx)

ansarid, 2019-03-13:
Google Speech Recognition Library:
`sudo pip3 install SpeechRecognition`

PyAudio required to get audio from microphone.
`sudo pip3 install pyaudio`

Code:
```
#!/usr/bin/env python3

import speech_recognition as sr

# get audio from the microphone
r = sr.Recognizer()

while 1:

	with sr.Microphone() as source:

		print("Speak:")

		audio = r.listen(source)

		try:
			print("You said " + r.recognize_google(audio))
		except sr.UnknownValueError:
			print("Could not understand audio")
		except sr.RequestError as e:
			print("Could not request results; {0}".format(e))
```

This library is super buggy and probably does not work well enough to be useful to us.
**Further testing is required.**

closed 2019-03-13 by ansarid