scuttlerobot/SCUTTLE #20: SICK lidar - find USB com settings

dmalawey, 2019-04-02:
1) Determine the communication method for the TiM561 over USB. \
e.g. if the communication is "serial" then what is the baud rate?\
Otherwise, determine the method, default IP address, port number, etc.\
2) share the spreadsheet that breaks down the "telegram" from the lidar.

dmalawey, 2019-04-02:
Attachments - example telegrams
[sample telegram.txt](https://github.com/MXET/SCUTTLE/files/3032186/sample.telegram.txt)
[telegram_translator_ver1.0.xlsx](https://github.com/MXET/SCUTTLE/files/3032187/telegram_translator_ver1.0.xlsx)

dmalawey, 2019-04-03:
from a SICK rep:

1) I found a surprisingly-detailed "how-to" in our internal support portal.  I attached the PDF and ZIP at the bottom that shows the process.  My research suggests that telnet programs can not readily connect to USB devices until a driver has "translated" the USB connection into some type of serial connection (COM1, COM2, etc).  The guide seems to show that the ROS driver can be used standalone (there's a link) but I'm not enough of an expert to say for sure.  Otherwise, there's a load of information in the PDF and ZIP to basically write your own driver...including example C code. 
2) This guide looks extremely useful (and succinct!) to get the core ROS functionality loaded on a Beaglebone running the most recent Debian release (Debian 9 "Stretch")

[USB_interface.pdf](https://github.com/MXET/SCUTTLE/files/3041128/USB_interface.pdf)
[TIM_Linux.zip](https://github.com/MXET/SCUTTLE/files/3041129/TIM_Linux.zip)

dmalawey, 2019-04-04:
from an MXET senior:

here is a python program which will open and read from the USB port as a serial connection. It is commented fairly heavily with explanations of what is happening as well as links to documentation for several of the imports. 

(after downloading, change the file extension to .txt)
[USBAsSerial.txt](https://github.com/MXET/SCUTTLE/files/3044561/USBAsSerial.txt)

ansarid, 2019-04-10:
Trying to build `TiM3xxLibUSBExample.c`
```
debian@scuttle:~/mxet_dev/LIDAR/TIM_Linux$ gcc TiM3xxLibUSBExample.c -o TiM3xxLibUSBExample
TiM3xxLibUSBExample.c:15:20: fatal error: libusb.h: No such file or directory
 #include <libusb.h>
                    ^
compilation terminated.
```
I tried installing libusb using apt-get and installed different versions of libusb which did not resolve the issue. I also built libusb from source which did not work.

ansarid, 2019-05-03:
I've now written [a simple library to use the TiM561 LIDAR over USB](github.com/ansarid/pytim5xx).

closed 2019-05-03 by ansarid