scuttlerobot/SCUTTLE #23: X11 Forwarding from BBB Issue

d-belleza, 2019-04-04:
When trying to ssh using X11 forwarding, we get an error message:
X11 forwarding request failed on channel 0

We have tried to modify the sshd_config file by changing the lines to:
X11Forwarding yes
X11UseLocalhost no

We do not know to how allow X11 forwarding to show the captured frame from OpenCV.

ansarid, 2019-04-04:
You may not have the X11 server installed.
Try ```sudo apt-get install xorg```.

d-belleza, 2019-04-04:
Trying to install xorg and run into the issue of
`You don't have enough free space in /var/cache/apt/archives/.`

ansarid, 2019-04-05:
Please try:
```
  cd /opt/scripts/tools
  git pull
  sudo ./grow_partition.sh
```

dlnash96, 2019-04-08:
The solution you posted for increasing the partition size worked. However, we now get the error message: 

> "X11 connection rejected because of wrong authentication."

This happens when using a very simple script that calls xeyes on the beaglebone. 

`import os`
`os.system("xeyes")`

[image]

ansarid, 2019-04-13:
You cannot call a gui application using os.system() on a remote server because though X11 works on your beaglebone it will not forward the gui through the X11 display.

closed 2019-04-13 by ansarid