scuttlerobot/SCUTTLE #26: X11 "Connection Refused" error when attempting to run ColorCalibrate.py

inioluwa4, 2019-04-22:
[image]

X11 forwarding is enabled in Putty
Xming server is active.

dmalawey, 2019-04-24:
@inioluwa4 please include this infomation.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, include code to help explain your problem.

**Platform information**
Please let us know the debian date of the image you are runnning.

dmalawey, 2019-05-02:
No details were provided for this issue.  Closing now.

closed 2019-05-02 by dmalawey

ansarid, 2019-05-02:
@inioluwa4 That program opens a GUI window that is called from the cv2.imshow() function. This program will work using X11 forwarding which I assume is what you are trying to do. You need to use a X11 capable SSH client like MobaXterm so that the GUI window can be forwarded to you. The terminal in Cloud9 does not have this capability which means running any programs that launch a GUI through the Cloud9 terminal will fail.

Also please copy and paste error outputs instead of taking a screenshot of the issue. This helps other users with the same or similar issues find the solutions to their issues more easily.