scuttlerobot/SCUTTLE #27: Servo motor control problem

mortiz139, 2019-04-24:
Using two servos to control a robotic arm. I am currently trying to get one servo running, but as soon as I run the code nothing moves. My code is essentially based off what was provided in this github. I also tried running rc_test_servos too, and it still not move. 

I keep on getting the "Installing signal handlers" warning. I have already done the two updates. 
sudo apt-get update
sudo apt-get install librobotcontrol

My code is posted below 

# import python libraries
import time, math
import getopt, sys

# import rcpy library
# This automatically initizalizes the robotics cape
import rcpy 
import rcpy.servo as servo
import rcpy.clock as clock

# set state to rcpy.RUNNING
rcpy.set_state(rcpy.RUNNING)

# Create servo

clck = clock.Clock(baseservo, 0.02)

try:

	# enables all servos
	servo.enable()

	# start clock
	clck.start()

	# keep running
	while rcpy.get_state() != rcpy.EXITING:

		baseservo.set(1.3)
		

except KeyboardInterrupt:
	
	pass

finally:

	# stop clock
	clck.stop()

	# disable servos
	servo.disable()

	# say bye
	print("\nBye Beaglebone!")

ansarid, 2019-04-24:
The "Installing signal handlers" is just a warning, they can be ignored.

How are you powering your Beaglebone?

mortiz139, 2019-04-24:
I was using the power brick that was given to us to power it.

ansarid, 2019-04-24:
@mortiz139 Do you get any servo movement from running the command `sudo rc_test_servos -s 1.5`?

dmalawey, 2019-04-24:
@mortiz, next time you post code, use the "insert code" button so the formatting is understandable.

You can grab the servo example 1 from the python basics folder, and that will run 1 servo.  Then use the code below to expand that to two servos, plugged into channel 1 and 2.  

If this works, take some time to review the difference between the basic example and the one here, so you can understand what's going on.

```
# This example moves two servos to the position specified by "duty," then reverses, then repeats.   
# If the servo is a continuous type, the duty will set the speed instead of the position. 
#To use this example, your servos must be connected to the first and second ports.

# import python libraries
import time, math
import getopt, sys

# import rcpy library
# This automatically initizalizes the robotics cape
import rcpy 
import rcpy.servo as servo
import rcpy.clock as clock	# For PWM period for servos

# defaults
duty = 1.5	# Duty cycle (-1.5,1.5)
period = 0.02	# Set servo period to 20ms
channel = 1	# Which channel (1-8), 0 outputs to all channels
channel2 = 2

# set state to rcpy.RUNNING
rcpy.set_state(rcpy.RUNNING)

srvo = servo.Servo(channel)	# Create servo object
srvo2 = servo.Servo(channel2)

clck = clock.Clock(srvo, period)	# Set PWM period for servos
clck2 = clock.Clock(srvo2, period)

try:

	# enable servos
	servo.enable()		# Enables 6v rail

	# start clock
	clck.start()		# Starts PWM
	clck2.start()

	# keep running
	while rcpy.get_state() != rcpy.EXITING:

		srvo.set(duty)	# Set servo duty
		srvo2.set(duty)
		time.sleep(2)
		
		srvo.set(-duty)
		srvo2.set(-duty)
		time.sleep(2)
        
except KeyboardInterrupt:
	
	pass

finally:

	# stop clock
	clck.stop()

	# disable servos
	servo.disable()

	# say bye
	print("\nBye Beaglebone!")
```

closed 2019-04-24 by dmalawey

mortiz139, 2019-04-24:
I am off campus grabbing lunch real quick, but I will try it once I’m back and let you know.