scuttlerobot/SCUTTLE #28: input/output error & overall strange behavior in Shell

dmalawey, 2019-04-26:
Beaglebone is not cooperating in the Shell session.  How can we troubleshoot this?

**Steps to reproduce the problem:**
Upon loggin into the beaglebone, there is a "locking failed" message. 
Upon further commands such as dmesg, there is an Input/output error for any command.

**Shell commands and output here:**
```
debian@192.168.8.1's password:
Last login: Sun Feb  3 21:03:56 2019 from 192.168.8.65
zsh: locking failed for /home/debian/.zsh_history: read-only file system: readin                        g anyway
➜  ~ dmesg
zsh: Input/output error: dmesg
➜  ~ sudo dmesg
sudo: unable to open /var/lib/sudo/ts/debian: Read-only file system
[sudo] password for debian:
sudo: unable to execute /bin/dmesg: Input/output error
➜  ~ dmesg
zsh: Input/output error: dmesg
➜  ~
```

ansarid, 2019-05-03:
Is this an issue you have often? It looks like something that would happen if your SD card popped out while the beaglebone was in use. Did rebooting solve this issue?

newday1, 2019-07-08:
I have this issue too. My os is fedora 30.  Rebooting can't sovle this issue

closed 2019-08-06 by ansarid