scuttlerobot/SCUTTLE #29: ADC initialization error?

mortiz139, 2019-04-26:
I am currently just trying to run my servos. Nothing is plugged in to the Beagle Bone at the moment, but once I run it, this is the error I receive. 

debian@scuttle:/var/lib/cloud9$ rc_test_servos -c 1 -p 0.0
ERROR in rc_adc_init, failed to open iio adc interface
Failed to write to 'state.settings'. EROFS: read-only file system, open '/var/lib/cloud9/.c9/state.settings' .

dmalawey, 2019-04-26:
This trial was repeated on a different beaglebone with the same SD card, and the code executes with no error, and servo runs properly.  Does this mean the blue has a problem?

ansarid, 2019-04-26:
@mortiz139 try running it with sudo

ansarid, 2019-05-03:
Inactive issue for what seems like a lack of knowledge instead of an issue. Closing.

closed 2019-05-03 by ansarid