scuttlerobot/SCUTTLE #33: Self_installer.sh does not complete

dmalawey, 2019-08-30:
The self_intaller does not finish it's job of installing packages and many errors follow.  

An example of the output is shown in this gist:  [problems with self_installer](https://gist.github.com/dmalawey/548086df8b3069f73360ee25d0f626f2)

dmalawey, 2019-08-31:
A resolution to this problem has been generated (with help from Ahmad B.)  We will reopen the issue if this fix is not successful for other users.

**Please note the fix:**
The self-installer starts by expanding the partition containing the image on the SD card, so there is room for additional installations of packages.  After this "growing" of the partition we need to reboot so the machine recognizes the larger partition space.

*The new self_installer has been modified to have two parts.  After the first part, a reboot is commanded. *

**Steps for the new process:**

- Reflash the SD card
- Reconnect to wifi using the tamulink_wpa process or using [this process](https://gist.github.com/dmalawey/e4be9f3b5beba8ed49868065935127b7) at home.
- Run the self_installer [part 1](https://github.com/MXET/SCUTTLE/blob/master/software/scripts/self_installer_part1.sh)
- After the machine reboots, reopen a terminal and run self_installer [part 2](https://github.com/MXET/SCUTTLE/blob/master/software/scripts/self_installer_part2.sh)

ReinhardtTanner, 2019-09-02:
I was able to complete self installer part 1, however I keep losing connection in cloud9 during step 2.

This is being done on my network at home because TAMU won't even let me do self installer part 1 for some reason.
I'm getting a read only error on the TAMU network.

An image is attached displaying the error.
[image]

dmalawey, 2019-09-02:
@ReinhardtTanner, you flashed using the very top version on the beagleboard.org/latest webpage, correct?  The name is **Debian 9.5 2018-10-07 4GB SD IoT**

The version called "flasher" may cause a problem since it attempts to write to the Blue's memory directly and subsequently will boot from onboard memory, limited to 4gb.

ReinhardtTanner, 2019-09-02:
Let me redo the process so I can narrow down a single issue.

ReinhardtTanner, 2019-09-02:
Just checked, looks like I used the version below it, the LXQT version. I
am currently downloading the correct file (IoT) and then I will re-flash.

ReinhardtTanner, 2019-09-02:
Hey David I reflashed the correct version from the BeagleBoard site, however I am still having the read only issue. I am currently on my home network. Below is the terminal text.

debian@beaglebone:/var/lib/cloud9$
debian@beaglebone:/var/lib/cloud9$ cd ~
debian@beaglebone:~$ sudo bash self_installer_part1.sh
sudo: unable to open /var/lib/sudo/ts/debian: Read-only file system
[sudo] password for debian:
debian@beaglebone:~$

dmalawey, 2019-09-04:
I think the next step is to learn what's going on with the space on the drive.  Please run commands from this video and output info on your boot drive. 

https://youtu.be/SzzNJBzTZss

@ansarid  please add other useful troubleshooting questions to check on.

ansarid, 2019-09-04:
@ReinhardtTanner At exactly which step in the lab does this issue arise or become apparent?

ReinhardtTanner, 2019-09-18:
Ok so the issue was resolved. It seemed like it was a problem with the board as the WiFi and connection to the Beagle from computer frequently dropped upon attempted execution of the install files. The two install files installed flawlessly once a new board was obtained. Keep in mind the SD card remained unchanged with the previous flash version.

closed 2019-09-25 by ansarid