scuttlerobot/SCUTTLE #36: Add a licence for a project?

msadowski, 2019-09-20:
Hi! Thanks for making this project and sharing it!

I'll be featuring it in the next issue of [Weekly Robotics](https://weeklyrobotics.com/). I've noticed that there is no project licence anymore in the project (there was a commit that removed it). As far as I know projects behind licence default to "All Rights Reserved" meaning that theoretically no one should use your work.

dmalawey, 2019-09-20:
Hi Mateus,

I am not savvy with license terms but once I can review our current
situation I’m going to assign a license such that copies of our work should
be required to mention the original authors, or Texas A&M department of
ETID, (engineering Technology department).

I will humbly ask that you be sure to do this in your article.  Thank you
for being forthcoming on this.  Let me know if you need any details to
share in your literature.

Cheers,

David Malawey
Texas A&M University
Malawey@tamu.edu
Cell: 314-974-4479

msadowski, 2019-09-20:
Thanks David!

I'll be just linking to your main website. For finding a suitable licence I can recommend taking a look at https://tldrlegal.com/, they make good summaries of licences.

Also you can take a look at the database I've created during my work on the newsletter, where I specify licence for every project that has it: https://github.com/msadowski/awesome-weekly-robotics, that way you can see what kind of licences other projects use.

Once again thanks for sharing this project!

ansarid, 2019-09-25:
https://github.com/MXET/SCUTTLE/blob/master/LICENSE

closed 2019-09-25 by ansarid