scuttlerobot/SCUTTLE #43: name 'lidar' is not defined

jessaggie, 2019-11-03:
Lidar is not defined even when it is defined. 
We have verified the system recognizes it. 
We are using the TiM781 lidar. 
After playing around we believe it is a read/write issue but could be wrong. 

Running the L1_lidar file gave us this error. It is in the same location as the pysicktim file. 
The vendor id and product id are also exactly the same. 

`debian@scuttle:~/basics/pysicktim$ python3 L1_lidar.py
Traceback (most recent call last):
  File "L1_lidar.py", line 48, in <module>
    lidarData = polarScan(54)
  File "L1_lidar.py", line 19, in polarScan
    lidar.scan() #take reading
  File "/home/debian/basics/pysicktim/pysicktim/pysicktim.py", line 351, in scan
    send('sRN LMDscandata')
  File "/home/debian/basics/pysicktim/pysicktim/pysicktim.py", line 196, in send
    lidar.write(2|usb.ENDPOINT_OUT,"\x02"+cmd+"\x03\0",0)
NameError: name 'lidar' is not defined`

ansarid, 2019-11-03:
@jessaggie It looks like the library was not installed correctly. Please read the instructions on the pysicktim README.md. Also please post any issues regarding the pysicktim LiDAR on the pysicktim repository issues tab.

dmalawey, 2019-11-03:
@jessaggie are you using the L1_lidar.py file that I updated yesterday?   Please confirm you have the latest.

jessaggie, 2019-11-03:
We were using the latest L1_lidar.py file.  Ran installation again and it worked this time. Thanks!

closed 2019-11-03 by jessaggie