scuttlerobot/SCUTTLE #47: all around Beagle Connection issue

jessaggie, 2019-11-14:
We turned off the wifi in connman with the command connmanctl disable wifi and can no longer see anything when hard wired in through MOBA XTerm and through SSHing in.

jessaggie, 2019-11-14:
We were able to reenable the wifi access point through the serial port when hard wired in. When doing a normal boot after we were able to get on the beagle.

closed 2019-11-14 by jessaggie

jessaggie, 2019-11-14:
Plug beaglebone into the computer through the USB port. Open up Moba Xterm and start a new terminal selecting serial connection. Select usb/uart and set the baud rate to 15000. After this run the commands below. 
`connmanctl`
`agent on`
`enable wifi` (this is the most important one)
`scan wifi`
`services ` (should display an SSID address)
`connect (insert SSID address from above)`

dmalawey, 2019-11-15:
I'm glad it was a quick fix in the end!