scuttlerobot/SCUTTLE #48: self installer issue with Debian 9.9

destantraeger, 2019-11-14:
Our SD card is flashed with version 9.9 of the appropriate Debian distribution and slf installer is no longer compatible. When part1 is run, it will throw this error and then reboot.

debian@beaglebone:/var/lib/cloud9$ cd ~
debian@beaglebone:~$ sudo bash self_installer_part1.sh
[sudo] password for debian:
10+0 records in
10+0 records out
10485760 bytes (10 MB, 10 MiB) copied, 1.01095 s, 10.4 MB/s
Already up-to-date.
Media: [/dev/mmcblk0]
sfdisk: 2.26.x or greater
Disk /dev/mmcblk0: 14.7 GiB, 15728640000 bytes, 30720000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x672a5d88

Old situation:

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *     8192 30719999 30711808 14.7G 83 Linux

>>> Created a new DOS disklabel with disk identifier 0xade4c5d5.
/dev/mmcblk0p1: Created a new partition 1 of type 'Linux' and of size 14.7 GiB.
Partition #1 contains a ext4 signature.
/dev/mmcblk0p2: Done.

New situation:

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *     8192 30719999 30711808 14.7G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or kpartx(8).
Syncing disks.
reboot

ansarid, 2019-11-14:
Please send the output of "blkid -h".

destantraeger, 2019-11-14:
debian@beaglebone:~$ blkid -h
blkid from util-linux 2.29.2  (libblkid 2.29.2, 22-Feb-2017)
Usage:
 blkid -L <label> | -U <uuid>

 blkid [-c <file>] [-ghlLv] [-o <format>] [-s <tag>]
       [-t <token>] [<dev> ...]

 blkid -p [-s <tag>] [-O <offset>] [-S <size>]
       [-o <format>] <dev> ...

 blkid -i [-s <tag>] [-o <format>] <dev> ...

Options:
 -c <file>   read from <file> instead of reading from the default
               cache file (-c /dev/null means no cache)
 -d          don't encode non-printing characters
 -h          print this usage message and exit
 -g          garbage collect the blkid cache
 -o <format> output format; can be one of:
               value, device, export or full; (default: full)
 -k          list all known filesystems/RAIDs and exit
 -s <tag>    show specified tag(s) (default show all tags)
 -t <token>  find device with a specific token (NAME=value pair)
 -l          look up only first device with token specified by -t
 -L <label>  convert LABEL to device name
 -U <uuid>   convert UUID to device name
 -V          print version and exit
 <dev>       specify device(s) to probe (default: all devices)

Low-level probing options:
 -p          low-level superblocks probing (bypass cache)
 -i          gather information about I/O limits
 -S <size>   overwrite device size
 -O <offset> probe at the given offset
 -u <list>   filter by "usage" (e.g. -u filesystem,raid)
 -n <list>   filter by filesystem type (e.g. -n vfat,ext3)

ansarid, 2019-11-15:
@destantraeger Please send the output of "lsblk".

ansarid, 2019-11-27:
Closing due to lack of response.

closed 2019-11-27 by ansarid