scuttlerobot/SCUTTLE #50: L2_color_target.py producing an error

destantraeger, 2019-11-30:
code will run properly for several iterations of the standalone loop but will fail the process the 'None' value at some point and produce the following error:

Traceback (most recent call last):
  File "L2_color_target.py", line 53, in <module>
    if target[0] == None:
TypeError: 'NoneType' object is not subscriptable

ansarid, 2019-12-02:
@destantraeger I don't see this code in the `L2_color_target.py` file. Is this custom?

ansarid, 2019-12-04:
@destantraeger Has this issue been solved? If I do not get a response within the next day I will close this issue. If this issue has not been resolved, please let me know. If I close this issue and the issue has not been resolved please let me know and I will reopen the issue.

closed 2019-12-11 by ansarid