scuttlerobot/SCUTTLE #57: Unable to track objects based off of RGB values on camera

connortolbert, 2021-10-29:
Howdy, I am using a Pi 3B and running the L2_track_target.py script and cannot get it to properly track RGB color combinations. I am using the image dated 2021-05-07 and have tested a multitude of different RGB combinations. I even tried testing them in different orders because I thought maybe they needed to be inputted in a different manner. I would like to be able to utilize this script to find and track objects based on RGB values and utilize the distance from the center printed out to follow a colored line made from tape.

dmalawey, 2021-10-29:
Please tell us what hardware you're using, including the camera.

Please tell us the behavior you DO GET since you DON'T GET "properly tracking RGB"

Then, please tell your setup steps in enough detail so someone could repeat the issue.  @ansarid do you still have the link that explains how to clearly state a problem?

connortolbert, 2021-10-29:
> Please tell us what hardware you're using, including the camera.
>  We are using the logitech c922 pro hd stream webcam and microsoft lifecam HD-3000 camera given in lab. 
> Please tell us the behavior you DO GET since you DON'T GET "properly tracking RGB"
>  it states "No target located" 
> Then, please tell your setup steps in enough detail so someone could repeat the issue. @ansarid do you still have the link that explains how to clearly state a problem?
I downloaded the L2_track_target.py script, added the variables shown above, and changed the pixel setup to 1920 x 1080 in L1_camera. Afterwards, I changed the get ratio function to divide by 1920 to get the correct values. Besides that all I did was change the variable being inserted to the function to test different colors in the L2_track_target.py script.

dmalawey, 2021-10-29:
Thanks @connortolbert now we are getting somewhere.  I think it's best to validate that your camera is producing the stream of jpeg images, before it gets pulled into python.  We have not tested the combo of Pi + Logitech C922 before.  

Take a look at [this guide](https://scuttlerobot.org/uploads/SCUTTLE_Operation_Guide.pdf) in section 2.1, Module 1.  This manual is copyrighted and we are still adjusting it so don't distribute.

connortolbert, 2021-10-29:
> before. ‍ ‍ ‍ ‍ ‍ ‍ ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> This message came from outside your organization.
> ZjQcmQRYFpfptBannerEnd

Okay, thanks for the help!

closed 2021-11-09 by ansarid