scuttlerobot/SCUTTLE #58: Getting RCPY and PYCNTRL to Install Appropriately...

silver2row, 2021-11-09:
Hello,

Seth here. W/ GCC-10 on Bullseye, rcpy and pycntrl do not get installed appropriately. So, I am currently working on this issue.

...

If anyone has any recommendations, please reply.

Seth

dmalawey, 2021-11-15:
Hi @silver2row are you part of the beaglebone organization? We've had issues with rcpy for a long time and we were able to get some parts to work, while others were never resolved.  Are you here to help us? 😄

silver2row, 2021-11-15:
Hello...no. Sadly no. I am not affiliated w/ their organization.

Oh and by helping, you mean to alter source to work correctly w/ current changes in images, OSs, C standards, and C++ standards, maybe? 

I might be able to help a bit. 

@dmalawey , I may be able to make time to pitch in. I got their source to compile finally along w/ warnings in the Bullseye OS w/ kernel 5.10.x. 

Also, I have a listed error/warning pastebin paste on this link in librobotcontrol: https://github.com/beagleboard/librobotcontrol/issues/213 .

Seth

P.S. I take my time and move slowly, i.e. so, I may be working at this for some time. If you guys have a direction on what needs to transpire first, please let me know. I will pitch in when I can.

dmalawey, 2021-11-18:
@silver2row some of your references have gone over my head, but we need a person versed in embedded linux plus standards in C++ etc as you mentioned, for sure!

The most recent critical issue was related to the rcpy motor driving function just quitting on us, and a couple of my team members were teaching the Mobile Robotics lab when this happened multiple times so they have more details than I. If we can reproduce this issue I'll share it with you.

In 2020 and 2021 we focused on getting the Raspberry Pi fully integrated on SCUTTLE but next year I heard Beagle will release a new Blue which uses the new Texas Instruments TDA4VM.   At this time I hope to test out many new features they offer.

silver2row, 2021-11-19:
@dmalawey ,

Okay...no issue. Well, if you guys are going to use the new BBBlue when it arrives, nice. I am waiting on it too!

Seth

P.S. Anyway, if something happens and I get things configured correctly, I will reply.

silver2row, 2021-12-02:
@dmalawey ,

I have not figured anything out just yet. I think I am highly unskilled at getting someone else's lib. to work. I have resorted back to 4.19.x to make things work on my BBBlue for now.

Seth