scuttlerobot/SCUTTLE #60: i2cdetect not enabling

dmalawey, 2021-12-27:
From testing on 12/26:

~Fresh install of the raspberry pi image SCTL_PI_IMG_2021.07
~updates and upgrades
~delete SCUTTLE directory on the Pi
~git clone scuttlerobot/SCUTTLE
~try to run command "sudo i2cdetect -y -r l" and no result
~not sure what to try

```
SCUTTLE ~ ➜  sudo raspi-config
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.10.17-v7l+/modules.dep.bin'
modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.10.17-v7l+
SCUTTLE ~ ➜  sudo i2cdetect -y -r 1
sudo: i2cdetect: command not found
```

dmalawey, 2021-12-27:
I ran this and it looks like i2c detect works.  So, I think we need to add this install to the image. 

`sudo apt-get install -y python-smbus`

ansarid, 2021-12-27:
@dmalawey 

> ~Fresh install of the raspberry pi image SCTL_PI_IMG_2021.07

Please use the latest `11-06-2021_RPi_SCUTTLE_v0.21.1.img.xz` image. I'm about to create a new release(`12-27-2021_RPi_SCUTTLE_v0.21.2.img.xz`), so maybe wait for that one if you're going to reflash. I'm not sure what naming scheme `SCTL_PI_IMG_2021.07` is.

> ~delete SCUTTLE directory on the Pi
> ~git clone scuttlerobot/SCUTTLE

Are you just asking to remove the current SCUTTLE folder and clone it again from the latest source?

> ~try to run command "sudo i2cdetect -y -r l" and no result
> ~not sure what to try

I'll put a fix in the new image that installs `i2c-tools` which should resolve that issue. Not sure why that wasn't already included.

> I ran this and it looks like i2c detect works. So, I think we need to add this install to the image.
> 
> `sudo apt-get install -y python-smbus`

That might have worked, but that's not the correct solution. This will be fixed in the latest image.

> ```
> SCUTTLE ~ ➜  sudo raspi-config
> modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.10.17-v7l+/modules.dep.bin'
> modprobe: FATAL: Module i2c-dev not found in directory /lib/modules/5.10.17-v7l+
> ```

I was not able to reproduce this issue on the latest image.

dmalawey, 2021-12-28:
Do you have some log of what's installed in each image?  Can we produce this?

ansarid, 2021-12-28:
> Do you have some log of what's installed in each image? Can we produce this?

I have an installer script that installs all the required packages and tools.

closed 2022-02-10 by dmalawey