scuttlerobot/SCUTTLE #62: ipconfig, ifconfig not recognized

dmalawey, 2021-12-28:
2021.12 Fresh flash - ipconfig and ifconfig returns "Unknown command."

`/sbin/ifconfig -a` shows what used to show when I entered "ifconfig"

What to do?

ansarid, 2021-12-28:
Please use `sudo ifconfig`.

`ipconfig` is a Windows command and is not valid in Linux.

dmalawey, 2021-12-28:
> `ipconfig` is a Windows command and is not valid in Linux.

[image]

orly?

dmalawey, 2021-12-28:
Please see if you can review our long-standing linux notes for any other mistakes such as the missing sudo on the "iwconfig" entry.   Would appreciate it. 

https://github.com/scuttlerobot/SCUTTLE/edit/master/software/README.md

ansarid, 2021-12-28:
> > `ipconfig` is a Windows command and is not valid in Linux.
> 
> [image]
> 
> orly?

Yes. IP is a newer tool that's not at all related to ipconfig.

ansarid, 2021-12-28:
> Please see if you can review our long-standing linux notes for any other mistakes such as the missing sudo on the "iwconfig" entry. Would appreciate it.
> 
> https://github.com/scuttlerobot/SCUTTLE/edit/master/software/README.md

will do

ansarid, 2021-12-28:
@dmalawey Or we could drop our cheatsheet(or a lot of the commands on the cheatsheet) for the [better resources that already exist.](https://www.raspberrypi.com/documentation/computers/using_linux.html)

closed 2022-02-10 by dmalawey