scuttlerobot/SCUTTLE #64: LiDAR Data Inconsistent

lbbettinger, 2022-04-06:
Ranges of LiDAR Data points are coming in as 0 [distance angle]:
[image]
We were getting up to 10-15 0 values in a row earlier. This is while using a fully charged SCUTTLE battery pack. Using a bench-top power supply results in little to no errors.

Current (Temporary Wiring):
[image]
LiDAR is connected to the motor controller's 12V terminals. These terminals are then connected to the battery pack. The LiDAR power cable is currently fairly long since wire routing is still undecided. I suspect the wiring is the issue.

dmalawey, 2022-04-06:
OK here are some things you can check:

1. check the voltage of the battery.  the PIC has voltage/power meters that plug into anderson connectors.  Make sure in your trials the voltage is matched between battery and power supply
3.  check your surroundings - run your 2 trials with matching surroundings. (dont move the bot)  If the lidar strikes a reflective surface it won't get a value back.  Same if the distance is above 10m (see datasheet)
4. Please define "fairly long cable" - also I don't recognize this orange cable as we purchased a set of high-quality cables for all of the LIDARS in MXET 300 which you can probably borrow. 
5. Please define "little to no errors" - when you have errors what do they look like?
6. Please build a proper y-connector to split off your anderson power.  This configuration shown is not good practice.  We have videos for building splitters.

lbbettinger, 2022-05-02:
Shortening the cable and using Anderson Powerpole connectors resolved inconsistencies with the LiDAR

closed 2022-05-02 by lbbettinger